pace-neutrons/Horace #628: Logic error in smooth[_units] function argument parsing

nickbattam-tessella, 2021-03-18:
The sqw/dnd functions `smooth` and `smooth_units` call `smooth_dnd`. 

The argument parsing logic in this function means it's impossible to call with the `resolution` shape function.
```
if strcmp(shape,'resolution') && ~(ndim==2 && numel(width)==3)
    error ('Smoothing option ''resolution'' only available for 2D data sets with three width parameters')
elseif ~(isa_size(width,[1,ndim],'double') || isa_size(width,[1,1],'double'))
    error ('ERROR: argument ''width'' must be a scalar or vector with length equal to the dimensions of the dataset')
```

The first if requires that the data is 2d with 3 width parameters (for resolution), the second if requires that that width array is an array of size `[1, 2]`, or scalar for all shape functions in the 2d case.

cmarooney-stfc, 2023-09-19:
@cmarooney-stfc check if this code is still present

cmarooney-stfc, 2023-09-22:
@abuts @tgperring Toby, Alex - I was charged to see if this problem still exists - it does. I have proposed a fix above, but need a smoothing expert to verify - could you both have a look please? thanks Chris