pace-neutrons/Horace #631: Standardise identifiers for common errors

oerc0122, 2021-03-23:
Currently, there is no written standard for any standard identifiers which should be used in PACE for common error types. 

Suggest expanding [ADR 21](https://github.com/pace-neutrons/Horace/blob/master/documentation/adr/0021-errors-and-warnings.md) to include some common error types and the identifier which should be used with them and a general style for error identifiers (lower_snake seems to be most common in PACE, but matlab internals use PascalCase).
From a brief search, the following were common:
- `invalid_argument` - Any bad arg: invalid, wrong type, unexpected, etc.
- `invalid_arguments` - Suggest removing in favour of singular as more general
- `file_not_found`
- `io_error` - Generalised to be favoured over FileNotFound?
- `not_implemented` - Abstract/Virtual functions
- `array_mismatch` - Ranks or sizes 
- `runtime_error` - Could be anything?

nickbattam-tessella, 2021-03-23:
I think this is a new ADR rather than extension to the existing one

hsautessella, 2021-03-23:
We have a few of these standards now scattered over several ADRs - which is fine, but would make it a real pain for any newcomer on the project to get up to speed.

Would it be useful to condense these into a style-guide at this point? (If time allows)

nickbattam-tessella, 2021-03-23:
I think the a style-guide is a different thing to the ADRs: the ADRs record the decisions taken and the reasons for them. Each has a smaller scope than "all of the Horace codestyle".

The style-guide is a good idea and should represent the the consequence of all these decisions

oerc0122, 2021-03-23:
A style guide is certainly on the cards at some point, there was even a plan to have discussions with Euphonic and SpinW folks and try to develop a PACE style guide. Though when we'll be ready to do that, I don't know.

closed 2021-04-07 by cmarooney-stfc