pace-neutrons/Horace #724: Non-deterministic results from Tobyfit

rebeccafair, 2021-09-30:
I have recently found an issue with seemingly unrelated changes in Horace causing [this Tobyfit test](https://github.com/pace-neutrons/pace-integration/blob/4d304883a59eeb44708775a60395aab43817f1de/EuphonicDisp2SqwTest.m#L44) in pace-integration to fail.

Sometimes setting the random number seed at the beginning of the test isn't enough, as changes in Horace can cause the code path to change, changing the random number generator state to change by the time it gets to Tobyfit. e.g. the PixelData constructor [calls randi to generate the object ID](https://github.com/pace-neutrons/Horace/blob/3b12d6f7d7fcc771228dc27bae4b1693ad991c08/horace_core/sqw/PixelData/%40PixelData/PixelData.m#L321), so if the number of PixelData objects created changes, the Tobyfit results will also change.

Validating and regenerating test data every time something changes doesn't seem sustainable, so maybe there is a better solution.

cmarooney-stfc, 2021-11-01:
@oerc0122 Assigning this to you as it's Tobyfit and looks like a random number issue and you've been here! But happy to discuss alternatives

oerc0122, 2021-11-03:
Suggestion would be to use comparison against sigma (given by fit) as well as abs/rel tol, if they're within `N` sigma accept the result. 

Example from new test PR in progress

```matlab
function equalToSig(this, fp1)
 % Tol as sig, abs, rel
     tol = this.tol;
     var_name = inputname(2);
     fp2 = this.getReferenceData(var_name);

     sig1 = get_sig(fp1);
     sig2 = get_sig(fp2);

     tol_from_err = abs(tol(1)) * sqrt(sig1.^2 + sig2.^2);
     tol = {max(tol_from_err, tol(2)), tol(3)};
     message = '';

     for i=1:numel(fp1.p)
	try
	    assertEqualToTol(fp1.p(i), fp2.p(i), [tol{1}(i), tol{2}])
	catch ME
	    switch ME.identifier
	      case 'assertEqualToTol:tolExceeded'
		message = [message, newline(), '(', num2str(i), ') ', ME.message];
	      otherwise
		rethrow(ME)
	    end
	end
     end

     if ~isempty(message)
	error('assertEqualToTol:tolExceeded', message)
     end

end

function p=make_vec(pin)
    if ~isempty(pin)
	if iscell(pin)
	   p=cell2mat(pin);
	else
	   p=pin;
	end
    else
	p=[];
    end
end

function sig = get_sig(fp)
    if isfield(fp,'sig')
	sig=fp.sig;
    else
	sig=[];
    end

    if isfield(fp,'bsig')
	bsig=fp.bsig;
    else
	bsig=[];
    end
    sig = [make_vec(sig), make_vec(bsig)];
end
```