pace-neutrons/Horace #764: Horace GUI does not interpret gui strings properly

abuts, 2022-01-17:
Current Horace GUI reads input strings and converts them into cellarrays instead of vectors. This is the bug, which probably  been introduced with Matlab changing some formats. 

Should have trivial fix. The worry is why the bug had not been  noticed for so long.

mducle, 2022-01-17:
So David asked me about this too... it's because the original code used `strread` which is deprecated, and Jacob did a bulk change [here](https://github.com/pace-neutrons/Horace/commit/9bd71c44d487be419fd5ae26d509058f1adf55cf#diff-6add8ec4637494528bdd0b64a010eaf6fcdc77026fafbfbc4db0ae8b3b2ce5d8) which replaced all `strread` with `textscan`. Unfortunately, `textscan` has a different behaviour to `strread`... it's seems that the GUI was never tested after this change (despite several unrelated commits afterwards being made). This is because the changes to the GUI were made (and probably tested) on the `rel_3_5` branch whilst the `strread`/`textscan` changes were made on master, and it seems like the GUI was never tested on master before it was merged... and never tested afterward when `rel_3_6` was made from `master`.

I think we need to instigate a manual testing protocol before release like Mantid does...

abuts, 2022-01-17:
> 
> I think we need to instigate a manual testing protocol before release like Mantid does...

Thanks god -- this explains the issue. I've fixed it. Nobody tests GUI-s -- this is the problem. We need to spend some time on making tests for them -- general care for GUI is necessary -- no much time is needed but benefits will be obvious.

closed 2022-03-03 by abuts