pace-neutrons/Horace #768: Erroneous deletion of routines

russellewings, 2022-01-31:
Several used (by me!) routines have been deleted between Horace_3_2 and Horace_git (on IDAaaS, as of 31/1/22). Examples include run_inspector and replicate sqw methods. These are documented and useful commands so need to be reinstated. Alex suggests that work may be needed to ensure that they work correctly with the new pixel class

cmarooney-stfc, 2022-01-31:
@russellewings Hi Russell. Please could you be explicit about the full path of the routines you know have disappeared. Your text suggests you expect that more have disappeared than you know. If you could add detail that would be good. Is this the standard Horace version on IDAaaS or one that you have installed yourself? In either instance, could you say what the Horace and Herbert version numbers are displayed by horace_on - I'm not expecting Horace_git.
@abuts If you know what commit deleted these routines, could you add the info to this ticket please.

abuts, 2022-01-31:
@cmarooney-stfc; @russellewings  -- Chris, Russell have explained me the details of the issue. I well understand what is necessary. 
Will start dealing with this tomorrow

mducle, 2022-01-31:
For the record, these are the files removed between release `v3.5.3` and release `v3.6.0` when the new SQW class was merged in:

https://gist.github.com/mducle/660f7eb894c2aefb4c4710200ef76572

There are 645 files but most of them were not really removed, just moved elsewhere (e.g. the `@dnd/*` functions were moved into `@SQWDnDBase/`. A reduced list which removes any files which has the same name elsewhere is here:

https://gist.github.com/mducle/077e57dfb5d09dad88f974593596a57d

and has 201 files...

abuts, 2022-01-31:
majority was not removed but moved to base classes, working on parents rather then repeating the same implementation on children. But some of them were erroneously removed indeed.

closed 2022-03-09 by abuts