pace-neutrons/Horace #798: in memory and in file symmetrisation define different image range

abuts, 2022-05-05:
New memory symmetrisation routine calculates final symmertisation regions by transforming current image bounding box using symmetrisation transformations. 
The file-based symmetrization routine does this with every contributing file and does not combine these ranges together. As the result, the final range for the symmetrized image for filebased data is different from the range, obtained when all ranges are calculated in memory in one go, despite the ranges of the contributing pixels and general symmetrized image looks similar. 
(see `gen_sqw_accumulate_sqw_tests_common:test_test_gen_sqw_sym`). This is the bug that should be fixed.

oerc0122, 2023-06-20:
Review with new symmetrise

cmarooney-stfc, 2023-09-19:
@oerc0122 discuss prioritization on return with @cmarooney-stfc

closed 2025-12-16 by abuts