pace-neutrons/Horace #801: test_tobyfit_resfun_2:test_projaxes_3  fails on aspect ratio

abuts, 2022-05-16:
test_tobyfit_resfun_2:test_projaxes_3  fails on checking figures aspect ratio. Despite the images look reasonable and slightly skewed, the aspect ratio, returned from the graphical objects comes as [1,1] while it should be something like [0.3820    0.7958] (and the image looks like it has aspect  [0.3820    0.7958].) Need to check with Toby what it means and how this failure could appear

tgperring, 2023-01-31:
This bug is a symptom of a wider problem. The initialisation of the sqw object that is done in  ..\horace_core\algorithms\resolution_plot prior to making the call to sqw/resolution_plot which does the actual plotting has a problem. Th general case of initialising the plot axes with optional input argument has been lost in the conversion to the new sqw structure since the release version Horace 3.6.2. The result is that the plotting axes are not always correct; also I beliecve this is responsible for the problem with enquiring of the plot aspect ratio.

For example, in test_projaxes_3 the two plots that are created should differ only in the second being the first rotated by 90 degrees (confirmed by running in 3.6.2 on 31/1/2023). However, now the aspect ratios are incorrect: the plots should be created by having the same number of cm on the screen along x and y axes corresponding to the same number of Ang^-=1. This is why the aspect ratio e.g.  aspect2 = get(gca,'DataAspectRatio') shoudl have aspect2(1:2)  = [0.3820    0.7958]. It now in the master as of 31/1/2023 returns the value [1,1]. 

The other example is in test_differentDetectors, in which the plots for the N and S detectors should be at an angle to the x and y plot axes, but in the current master are not. This is because the setting of particular plot axes has been lost in the new sqw object initialisation.

abuts, 2023-06-21:
fixed through 
#892, commit # c622453b65365fd721ee3438e6666087d4a603d1