pace-neutrons/Horace #821: Relative (root) paths can be sanitised

oerc0122, 2022-07-07:
There are places in the code which are looking for e.g. the root path (the folder containing `(herbert|horace)_core`) or the `(herbert|horace)_core`s respectively. Currently, this is done (where necessary) through `fileparts(fileparts(which('herbert_init')))` this is not only confusing code, but also unnecessarily complex. 

To solve this we can add global variables `herbert_root`, `horace_root` and `root_path` to be initialised once in `herbert_init` and `horace_init` respectively (and likewise unset in `horace_off`) and reference these as needed.

oerc0122, 2022-07-07:
1. This is a bad habit.
2. Does this not clear other horace/herbert vars?
3. Is there a way to rectify this? I suppose I could attach it to the `horace|herbert_config`

mducle, 2022-07-07:
Yes, it's a bad habit but people _will_ use it. It clears all the variables but the Horace paths are still in memory so things like `cut_sqw` will still work. It's just that `ClusterMPI` (e.g. `gensqw` or configuring the parallel framework) and  `horace_off` will now fail...

> Is there a way to rectify this with extra persistence?

Not really. `clear all` really does clear everything.

The workarounds I can think of are:

* Write the path variables to a temporary file.
* Wrap the path variables in a small helper class which can still use the globals but if it detects the globals have been cleared will call `which('horace_init')` / `which('herbert_init')` to reset them (the path is not cleared)

closed 2022-12-12 by cmarooney-stfc