pace-neutrons/Horace #856: Bug when passing 3 sqw objects to tobyfit()

Oli-Rest, 2022-09-20:
When passing 3 sqw objects to `tobyfit()`, when setting the `local_foreground` property with `set_fun`, `pin` should be 3 separate cells of parameters, instead, the 3 `pin` values are set as the 1st, 2nd and 3rd parameters. This only seems to happen when 3 sqw objects are passed to `tobyfit()`, no issues when using 2,4, 6 etc… (Picture of bug included to help)
[image]

oerc0122, 2023-07-04:
This is a known thing and is partially documented:
```
        % Multiple functions
        % If the argument is a cell array with the same size as the array of
        % function handles, then assume that a cell array of parameter lists
        % has been provided.
        % Otherwise assume that a single parameter list has been given.
```
Solution is to use nested cell array:
```
kx = kx.set_fun(@slow_func, {{[250 0 2.4 10 5],@demo_FM_spinwaves,10^-1}})
```
Whether this **should** be the case is another matter.

oerc0122, 2023-07-04:
Script to demonstrate (**N.B.** remove double brackets to incur error)

[temp.txt](https://github.com/pace-neutrons/Horace/files/11950480/temp.txt)