pace-neutrons/Horace #1025: Untested points in the code calling header_average

cmarooney-stfc, 2023-06-01:
While testing the uses of header_average (both the sqw method and the Experiment method), it turns out the following lines of code are not tested by our test suite. Tests should be reactivated or provided.

Breakpoint for @SQWDnDBase\private\parse_change_crystal_arguments_ is on line 62.
It is not actually clear that any of the example code in the file header actually refers to the elseif block around this line, and the file header should also be examined to see if it is fit for purpose. This is referred to in the code at.
Breakpoints for change_crystal are on lines  68, 73.
Breakpoint for sqw.change_crystal is on line 53.

Breakpoint for @sqw\private\cut_sqw_sym_main is on line 164.
Breakpoints for @sqw\private\cut_sqw_sym_main_single are on lines  64, 81.
These lines may be referred to by tests in test_sym_op that are currently skipped (ticket reference #805), so it would be good if they could be reinstated.

Breakpoint for MagneticIons.getFF_calculator is on line 47.

Breakpoint for algorithms\private\gen_sqw_check_sqwfile_valid is on line 59.
This should presumably be hit by the gen_sqw tests (accumulate_old_sqw, ~use_partial_tmp) but isn't

Breakpoint for sqw.header_average is on line 9.
Line 11 with nargout==1 is tested but not line 9 with the extra output argument ebings_all_same.

Breakpoints for sqw.lattice_parameters are on lines  20, 29.

abuts, 2023-08-16:
I believe, I've removed need in using header_average in new release, though it still can be used in compartibility mode. As this is not so urgen to use it in compartibility mode, I think it can be done post MVP

oerc0122, 2023-11-29:
Header average is used in the following places in Horace:
```
horace_core/admin/upgrade_file_format.m
68:            hav = exp.header_average;
horace_core/algorithms/helper_functions/is_legacy_aligned.m
27:hav = exp.header_average;
horace_core/algorithms/private/gen_sqw_check_sqwfile_valid.m
47:header_ave=exper_info.header_average();
horace_core/sqw/@SQWDnDBase/private/parse_change_crystal_arguments_.m
59:        header_ave=header.header_average();  % this gets the header for the first spe file that contributed.
horace_core/sqw/@sqw/lattice_parameters.m
20:h_ave=win(1).header_average();
29:    h_ave=header_average(win(i).experiment_info);
horace_core/sqw/@sqw/private/set_from_old_struct_.m
 58:                hav = header_average(ss.experiment_info);
 66:                hav = header_average(ss.experiment_info);
 96:        header_av = ss.experiment_info.header_average();
127:    header_av = obj.experiment_info.header_average();
horace_core/sqw/file_io/@sqw_binfile_common/get_sqw.m
142:header_av = exp_info.header_average;
horace_core/sqw/file_io/@sqw_binfile_common/sqw_binfile_common.m
136:            header_av = exp_info.header_average;
```