pace-neutrons/Horace #1033: fractional comparison is broken

abuts, 2023-06-09:
see test_pixels_equal-> test_using_fraction_argument_only_tests_fraction_of_the_bins

fails and debugging shows that fractions is not attempted.  

The test have been recovered merging master in #892. Its certainly not duplicated

oerc0122, 2023-06-09:
Compare `_test/test_sqw_class/test_equal_to_tol` and `test_pixels_equal`, relevant different different tests were merged across before removing the duplicates.

abuts, 2023-06-09:
following our conversation ,I need to add more. 
The previous comparison was comparting both pixels and distribution of pixels over bins. The second one is even more important as pixels in a bin can be distributed randomly, but it should be the same pixels. 
comparing just pixels does not make sense.

oerc0122, 2023-06-10:
That is handled by reorder which sorts pixels according to their inner indices. I explicitly disable it in the new test as otherwise the zeroed bins get pushed to the top of the list and the comparison fails

PR #1041 (2023-06-16): Restore bin masking on pix `equal_to_tol`
Restore masking bins on `fraction`al `equal_to_tol` between `PixelData` objects

Fixes #1033
closed 2023-06-16 by oerc0122