pace-neutrons/Horace #1051: Serializable may not handle objects with zero size

tgperring, 2023-06-19:
At the moment from_bare_struct_ has to catch the case of an empty input structure with no fields as meaning just return the template object. This is because we have used the case of such a structure as meaning 'not assigned' in the past. This is possibly poor design as a valid object to save/load is one with no size (i.e. an object array with size [0,0], [1,0] or [0,1] - in deed where prod(size(obj))==0. Serializable also interprets an empty object with fields in the same way - again possibly inconsistent design. This should be eliminated and the special cases of objects stored with empty structures handled in the particular cases where this has been done - notably the sqw object.

oerc0122, 2023-10-19:
Following review of problem, looks to be a significant change to the functionality of `serializable` as a whole to support it. Changing to post-construction as a result.

oerc0122, 2023-11-03:
Solved as part of #1366

closed 2023-11-03 by oerc0122