pace-neutrons/Horace #1146: Make object_lookup serializable

tgperring, 2023-07-23:
object_lookup needs to be converted to serializable. It is coupled to the Herbert utility methods that sort objects. The opportunity will be used to clear out the remnants of the obsolete I/O save and load protocol previously used by instrument classes (e.g. loadobj_private).

tgperring, 2023-07-26:
As part of these changes, the object_lookup tests as associated tests on classes created for those tests (boxClass and boxArrayClass) will be renamed so that the test functions themselves are more descriptive of what the tests are doing.

cmarooney-stfc, 2023-09-19:
@cmarooney-stfc close this when PR number is available

cmarooney-stfc, 2023-10-17:
@tgperring Toby, have you done this?

tgperring, 2023-11-22:
This task was completed in PR #1169

closed 2023-11-22 by tgperring