pace-neutrons/Horace #1213: `change_crystal` not respect deprecation

oerc0122, 2023-09-04:
`change_crystal` will error out if provided old-style correction rather than issuing warning and continuing.

cmarooney-stfc, 2023-10-05:
@oerc0122 Hi Jacob, can you add more detail here to enable someone to know where the failure is happening, what test it is in, what input needs to be provided - thanks

oerc0122, 2023-10-05:
This is not about tests, this is about user usage. 

The failure is in `change_crystal` and the deprecated state is highlighted by an explicit `if then error` rather than trying to coerce the parameters into a valid form if possible. It may not be, hence the vagueness of the issue.

oerc0122, 2023-10-18:
Determined to be difficult to maintain and fix.

closed 2023-10-18 by oerc0122