pace-neutrons/Horace #1293: bug in assigning w for non-orthogonal cut

abuts, 2023-09-26:
second cut in the script:
```
%sqw_file = 'test_monoclinic.sqw'
sqw_file = 'test_sqw_v4.sqw'

proj.u=[1,0,0];
proj.v=[0,1,0];
proj.uoffset=[0,0,0,0];
proj.type='rrr';
proj.nonorthogonal=false

E_cut=cut_sqw(sqw_file,proj,[0.025],[-0.1,0.1],[0.025],[-0.5,0.5],'-nopix');
plot(E_cut);
lx 0 5.2
ly -1.6 1.6
keep_figure

proj.nonorthogonal=true

E_cut=cut_sqw(sqw_file,proj,[0.025],[-0.1,0.1],[0.025],[-0.5,0.5],'-nopix');
plot(E_cut);
lx 0 5.2
ly -1.6 1.6
keep_figure
```
fails as w assigned in row 75 of 
`horace_core\sqw\coord_transform\@line_proj\private\projtransf_to_img_.m` routine is column vector instead of beeing row vector

cmarooney-stfc, 2023-09-26:
@abuts Hi Alex, no obvious problem with the PR but in the example above I thought cut_sqw had been replaced in Jacob's changes with just cut. Is it worth changing this?

abuts, 2023-09-28:
no -- cut sqw is still there for supporting old interface. I am not sure at all if cut is written as separate algorithm -- cut is the method on sqw object, buit if it is algorithm working with sqw file -- not sure about this. 

Actually it is there but cut_sqw is not deprecated. As all user scripts use cut_sqw, we should not deprecate it yet.

closed 2023-09-28 by abuts