pace-neutrons/Horace #1340: Additional work for detectors in Tobyfit code - x2 dependency on irun

cmarooney-stfc, 2023-10-19:
The Horace code has now been updated to create a compressed array of IX_detector_arrays on creation of the SQW object from the input nxspe files. These detectors are now used in the tobyfit_*_resconv_init code to initialise Tobyfit.

The resulting detector information in lookup_detector_table is the used in the tobyfit_*_resconv functions to perform the Monte Carlo calculation. idet has been added to the input to use the detector information.

However this does not completely cover all the data Tobyfit uses from the detectors. x2 is separately inserted into the lookup and is used to form dq_mat. It is processed by its idet indexing but the irun dependency is not used, and the x2 lookup is not created to provide irun dependency. This should be added to complete the detector info needed

tgperring, 2023-10-31:
I've looked at the code and concur. Additional functionality that is straightforward to implement is required to object_lookup/func_eval_ind, with some associated changes to the code of tobyfit_*_resconv_init, in particular constructing the object_lookup for the full set of IX_detector_array before the loop over sqw objects.

tgperring, 2023-11-23:
The code implemented in and merged from PR #1399 closes this issue.

closed 2023-11-23 by tgperring