pace-neutrons/Horace #1350: detector arrays are not restored to initial state after saving

abuts, 2023-10-23:
test: `~\_test\test_sqw_pageOpMethods\test_section:test_section_sqw_fb` saves detector array and then recovers them back

It does not touch these arrays in any way. 

Despite that, the filebacked object recovering the arrays differs from the original object, which means that the arrays are different. 
This is certainly the bug. 

In addition to that, `det_arrays` are not wired to `sqw.detpar` though this may be not a bug but the subject of the future work. 
Bug should be fixed. 

I've marked this issue in the test

cmarooney-stfc, 2023-11-06:
@abuts Looking at the test mentioned
(a) running this locally without the skip, it passes, and a comparison of *.experiment_info.detector_arrays for all sqws * mentioned says they are identical, because there are no detectors in these detector_arrays containers. This is because the detpars of these sqws are all empty; all their fields are empty. Please could you identify the differences that you observe, with particular reference to this description, and where you ran it. NB run by me with R2020b.
(b) for the comment that `detector_arrays` (presuming you mean this with `det_arrays`) are not wired to `detpar`, the code should now convert detpar to a unique_references_container of detector_arrays on loading the nxspes and unifying them. Thereafter, detpar is not needed and is just a useful reminder of the input - as, for instance, in debugging this ticket. Only the detector_array information should be used in the production code, which would be with Tobyfit. Please could you confirm that that is what you mean.

abuts, 2023-11-06:
If I disable row 58, which is there for fixing the test, the test fails. Skip there is just to identify the the row 58 have happened. 
I do not know why the test fails, I just know that I should recover what I've saved, even if it is different flavour of Matlab empty.

detpar should return whatever you have there, as it is what is accessed in production code. If it is container, its fine. It should behave as the structure before.

abuts, 2023-11-07:
If I disable row 58, test fails for me. I do not know how to reproduce anything else. If you think it works, remove this row and skip and this will be solution. If tests on Jenkins pass, I would believe the issue is fixed.

cmarooney-stfc, 2024-05-08:
Fixed by #1540

closed 2024-05-08 by cmarooney-stfc