pace-neutrons/Horace #1371: Current (old) binary operations were ignoring pixel order within bins

abuts, 2023-11-04:
Pixels may be randomly distributed within a bin. To perform binary operation between two sqw object's pixels properly, one-operand's pixel should interact with the other operand pixels with the same 4-D coordinate. 

This has not been implemented within the old binary operations.  If I try to implement such behaviour, some tests fail. Plus, maintaining the order requests resources and may be not necessary in some circumstances, e.g. if we are only interested in total signal and variance over the bin, we can add and extract them in any order.

Do we need to maintain order of pixels in operation and are there circumstances where this is not necessary?

cmarooney-stfc, 2023-11-15:
First fix added in #1303 (issue) fixed in PR #1370 see also merged PRs in 1303

abuts, 2023-11-17:
I believe, this changes resolve the issue. We've introduced new unsaveable property `sort_pix_in_binary_op` which controls sorting of pixels in binary operations. In the cases where sorting in binary operations is not important, this property can be set to `false`  to improve performance but by default it is `true` and pixel sorting occurs.

closed 2023-11-17 by abuts