pace-neutrons/Horace #1511: Image captions are invalid

abuts, 2024-01-29:
When cut image is expressed in A^(-1), axes still have inverse B-matrix element parts printed as caption of the axis. 
This is probably incorrect:
[image]

tgperring, 2024-02-06:
This behaviour is in fact correct. The axes still need to give the vectors that define the plot axes, which is done as [h k l], with the normalisation being such that in the example attached, the vector [0  0.454  0] has length 1 Ang^-1.

abuts, 2024-02-06:
x-axis on this plot is in A^(-1) and the unit of axis is 1A^(-1). This means that vector should be [0,1,0]

tgperring, 2024-02-06:
The vector in the captions along each of the axes for line_proj is always expressed in terms of hkl. Hopefully the following explains the meaning in the plot at the top of this ticket. The lattice in this case is cubic with lattice parameters all equal to  2.8526 Ang, so the lengths of a*, b* and c* are all equal to 2*pi/2.8526 = 2.2026 Ang^-1. Take the case of the vector b*: it is the vector [0,1,0] in hkl, and it has length 2.2026 Ang^-1. The vector along the same direction in hkl that has length 1 A^-1 is therefore the vector that is a factor 2.2026 shorter, that is the vector with value in hkl of [0, 1, 0] / 2.2026 = [0, 0.454, 0]. That is the basis vector in the caption to the x-axis in the plot at the top of this ticket. It means that if you are looking at a point for which the value along the x-axis is "1" , i.e. corresponds to zeta = 1, then the point has a component along the x-axis which in hkl is  [0, 0.454*zeta, 0] = [0, 0.454, 0]. If instead the point has a component zeta = 2, then the component along the x-axis in hkl is [0, 0.908, 0].

The same discussion applies to the y-axis of the plot. As an example of how to read the value of h,k,l for a point on the plot: Suppose the components of the point are (,4, 1) i..e. along the x-axis is -4, and along the y axis is +1. (This is a point close to the upper left corner of the middle detector bank at the left of the figure). The value along the x-axis is -4, and along the y-axis is +1. The point therefore has coordinates in hkl of: -4 * [0, 0.454, 0] + 1 * [0, 0, 0.454] = [0, 1.816, 0.454].

This explains the purpose of giving the basis vectors in hkl: it enables you to turn a coordinate of a point on the image axes i.e. the tickmark labels into a point in the reciprocal lattice of the crystal.

To label the x-axis as [0, zeta, 0] would tell the user nothing about the relationship of the plot axes to the crystal lattice.

abuts, 2024-05-08:
after some discussion has been decided to keep it as it is

closed 2024-05-08 by abuts