pace-neutrons/Horace #1583: smooth breaks consistency between pixels and image

abuts, 2024-02-28:
smooth algorithm works well on `dnd` object making image more pleasent to eye. 
When it is applied to `sqw` object, it chagnes image only so `recompute_bin_data` would return different result

This is not good as by default it is assumed that pixels and image are consistent between each other. 
`smooth` should be either prohibited on `sqw` or it should change `PixelData` to be consistent with image.

tgperring, 2024-02-29:
Perhaps the best thing is to output a dnd object, not an sqw object? That way we avoid the inconsistency problem, and then interpolating pixels, which would contravene the independence of the pixel data.

abuts, 2024-03-01:
Or may be just return `DND`. The question is do we need smooth pixels for any purpose

tgperring, 2024-03-04:
That's exactly what I meant.

abuts, 2024-03-04:
Do we need smooth pixels? If we just use resulting dnd in binary operations and plotting then certainly not. But if one wants something more advanced, like change pixels not by setting them all to image values but modifying according to some law (e.g. multiply them by some walue to have pixels avarage in cell equal to cell signal or smooth with something like option `-set_pixels` in remplicate) than it may be beneficial to modify pixels too. 

I do not see such requests at the moment but may be @tgperring does.

PR #1597 (2024-04-16): Make `sqw@smooth` return `dnd`
Fixes #1583
closed 2024-04-16 by oerc0122