pace-neutrons/Horace #1745: Energy cuts and Horace integration problem (Horace 4 and SpinW interface problem)

HyoukaChitandaEru, 2024-09-18:
Dear Horace Development Team,

I am encountering an issue when performing cuts on a SpinW model using Horace 4. Errors occur during the process (please see the attached figure), though everything works fine with Horace 3. I believe the issue may be related to Horace 4's new cutting algorithm.

To reproduce the bug, you can run Tutorial 10 ("Constant energy cut using Horace") from SpinW. The tutorial can be accessed via the following link: https://spinw.org/tutorials/10tutorial.

Best regards,
Zhaoyang
[image]

abuts, 2024-09-19:
It looks like the issue arises due to luck of integration testing as this type of constructor is not available in Horace 4. 

Can be easy added if necessary.

HyoukaChitandaEru, 2024-09-29:
Please add it. Thanks a lot!

abuts, 2024-10-04:
@HyoukaChitandaEru  -- to see anything in tutorial, these changes have to be merged, but row in 
tutourial 
`d3dobj = disp2sqw_eval(d3dobj,@sq.horace,{'component','Sxx+Syy+Szz'},fwhm0);`
should be modified:
`d3dobj = disp2sqw_eval(d3dobj,@sq.horace,{'component','Sxx+Syy+Szz'},fwhm0,'-all');`

Duc promised to do that.

closed 2024-10-07 by abuts