pace-neutrons/Horace #1762: horace_demo_script is not working due to reference to fit_sqw

henrikjacobsenfys, 2024-10-21:
This is issue is reported earlier in #1289 which is closed, but it's still present. I suggest changing the line

```
[wfit,fitdata]=fit_sqw(cc2a,@demo_FM_spinwaves,[250 0 2.4 10 5],[1 0 1 0 1],...
%     @constant_background,[0.05],[1],'list',2,'fit',[0.001 30 0.001]);
```

to
```
kk = multifit_sqw (cc2a);
kk = kk.set_fun (@demo_FM_spinwaves);

kk = kk.set_pin ([250 0 2.4 10 5]); %input parameters
kk = kk.set_free ([1 0 1 0 1]); %fitting parameters 1, 3 and 5

kk = kk.set_bfun (@constant_background); % set_bfun sets the background functions
kk = kk.set_bpin ([0.05]);   % initial background constant 
kk = kk.set_bfree ([1]);    % fix the background
[wout pout]=kk.fit();
```

abuts, 2024-10-21:
Dear @henrikjacobsenfys. Where did you find that? As far as I can see on IDAAAS, the suggested replacement have already been done.

Regards,
Alex

henrikjacobsenfys, 2024-10-21:
Hi @abuts 

Sorry for not being clear about this. It's in the Master branch of the repository: https://github.com/pace-neutrons/Horace/blob/master/demo/horace_demo_script.m 

Henrik

abuts, 2024-10-29:
fixed by Re #1763

closed 2024-10-29 by abuts