pace-neutrons/Horace #1768: USER ISSUE: lattice params a & b swap in coordinate transform

oerc0122, 2024-10-30:
For a triclinic/monoclinic system with scattering plane $u$=[110] $v$=[010] when transforming image for plot, the lattice parameters $a$ and $b$ appear to swap meaning when transforming coordinate frames. 

Details to follow following further chat.

abuts, 2024-11-05:
is projection options are `ppp` -- then yes, this is what is expected. Not sure about `rrr`. Happy to add `hkl` scalers like in spherical/cylindrical projections if this is the issue

abuts, 2024-11-07:
Swapping `a` and `b` means rotating x-axis by 3.26 degrees. As far as I understand -- around X-axis.
Script which does this is the modification of the script, provided by Vivian. 
The result looks as follows (last image of 3):
[image]

Still misaligned but original crystal is not well aligned too, so looks fine to me.
Should I do alignment for you? I saw the alignment script there -- is this result what you wanted to obtain?

abuts, 2024-11-07:
after alignment from lattice parameters [7.97,8.44,6.01] and gl=-3.281 brag-s postitions look correct. Crystal lattice parameters alatt = [8.3337,7.8805,5.9747]. This is how your crystal was oriented vrt the beam. You probably can do nothing about this:
[image]

abuts, 2024-11-08:
It is clear to me that the current Horace implementation is valid.

Your aligned your crystal in experiment so that beam is directed along
the line [1,1,0] where scales of the vector defining beam direction are
[2pi/8.44,2pi/7.997,0]/Norm. Horace beam is directed along axis q_x, so
this defines [8.44,7.997,...] lattice.

Currently you can not do anything about it. Note that your actual
lattice parameters are 10% smaller then the theoretical one.

If you oriented crystal in experiment at [2pi/7.997,2pi/8.44,0]/Norm,
lattice would be [7.997,8.44,...].

I think, the possibility to swap lattice parameters in your case could
be achieved by adding to Horace new option, which allows orienting beam
along any axis (x,y,z). This is a new feature which is doable but not
entirely trivial as may bring nasty surprises due to my limited
understanding of the problem. As the result, it needs to go along other
project priorities.

abuts, 2024-11-08:
closed acording to comment above

closed 2024-11-08 by abuts