pace-neutrons/Horace #1776: kf_sphere_proj

abuts, 2024-11-25:
If one have a look at  inelastic stattering signal in direct (rather than reciprocal) coordinate system, one can see clear elastic line signal at low energy transfer range. The signal can be easily approximated by double-side exponential function (see fig.2). 
[image]
fig.1
[image]
fig.2

The coordinates of axis in fig 1 are X-axis -- dE, Y-axis -- scattering angle in direct coordinate system. 

There are currently no way to produce such image in Horace, as $`\|Q\|`$ along Y-axis changes from 0 to $\|k_i\|$.

Similar image may be produced using $`\vec{k_f}`$ coordinates, namely $`\vec{k_f}=\vec{Q}-\vec{k_i}`$

This is the ticet to build spherical projection which would produce Horace image in $`\vec{k_f}`$ coordinates, namely $`\|k_f\|`$, $`\theta`$, $`\phi`$ coordinates where $`\theta`$, is the angle between beam and $`\vec{k_f}`$ directions.

cmarooney-stfc, 2025-02-19:
BLOCKED by #1152

abuts, 2025-03-27:
finalized by Re #1845. Further development/refactoring is possible on request

abuts, 2025-04-04:
Minimally addressed by Re #1845. Further development is possible on request