pace-neutrons/Horace #1873: add a standard signleton interface to `genieplot` class

abuts, 2025-05-20:
`genieplot` is a singleton which defines Horace/Herbert graphics. 

For some reason, it has specific to `genieplot`  interface to work with singleton in addition to the standart one. 

This ticket is to add standard signleton interface to the gieneplot one, for pieple with different experience to work with familiar interface to the class.

tgperring, 2025-05-20:
I agree that uniformity has merits. However, before making any changes, we should discuss what uniformity we want to aim for. I used a singleton template model (from Yair Altman) that is consistent with the way graphics handle properties are altered or retrieved, and is also very simple in its usage: you directly refer to the name of the singleton  (BTW, you can access all the properties as a structure to see what they are by using S=genieplot.get(),  again consistent with Matlab graphics handle objects).

The template I came across in Horace, `Singleton.m`, didn't give me much clue as to how it was used in practice. I downloaded the original version from the Matlab file exchange (Bobby Nedelkovski), which looks somewhat different, and it came with an example usage script that seemed a bit baroque, where you have to grab an instance before one can change a property, which personally I find a bit confusing. Perhaps the design pattern has more flexibility I haven't appreciated?

We should discuss the pros and cons. What I think would be a bad idea is to implement two different interfaces to genieplot.

abuts, 2025-05-20:
The change is better even for main reason:

If I type:
`genieplot.get` I got a structure, chaning which is useless. I need to remember to do `genieplot.set('propertyname',somethng)` to change something
if I do:
`genieplot.instance`, a class is returned. For this class you may use standard Matlab settings interface, tab autocompleteon etc.

Additional advantages and pro-arguments are:

1) there are bunch of other singletons in codebase, accessable through `class_name.instance`. Why somebody would want to use this singleton differently? 
2) old genie interface is there. If you want to use `genieplot.set('propertyname',somethng)` -- you welcome to do that. 
3) The standard interface code relies on Matlab's standard setters/getters. For old genie interface you need to write pretty complex custom glue code doing `obj.(property_name) = value` which looks uglier. It is there and is unit tested, so no problem of keeping it there, but if the question is on removing one interface, I would prefer the Matlab's standard one.

closed 2025-07-25 by abuts